sparky8512/starlink-grpc-tools #1: FR: Dockerize

neurocis, 2020-12-27:
Make this into a Docker container. I may be able to take this on and submit a PR.

sparky8512, 2020-12-27:
I'm not sure there is enough here to merit a Docker container, but if you think it would be useful, by all means, go ahead and implement.

One thing to note, though: I would consider the files generated by the protoc commands I pasted in the README to be derived from SpaceX copyrighted works, so unless and until SpaceX publishes the relevant .proto files with license terms that allow for redistribution, I will not personally be releasing a Docker image that includes those generated files.

neurocis, 2020-12-28:
I think grabbing the protoset and deriving the proto's can be done as part of the container's runtime initialization and not the build avoiding this proprietary works issue. And yes, the request was more useful to me and would share if I did tackle it. The work done so far is awesome! Cheers!

sparky8512, 2020-12-28:
OK, that sounds good. I'm going to leave this one with you, as I have no direct experience with Docker myself.

neurocis, 2021-01-05:
OK, hoping to get some cycles in-in the evenings this week, fingers crossed!

sparky8512, 2021-01-06:
FYI: I just pulled in a little code reorganization in preparation for adding a few more script variants. As a result, you will probably have to package up the starlink_grpc.py and/or starlink_json.py modules now, depending on which scripts you're targeting.

I'm not super happy with the module naming, and pylint really seems to hate my script naming conventions, but it works for now.

neurocis, 2021-01-10:
FYI - In looking at recent changes I am going to pivot and make the container more generic to just being capable of running the python scripts generically with the necessary dependencies.

neurocis, 2021-01-12:
I think the docker container will freeze here as it is just a generic host now regardless of the script being ran. Focus would move to allowing the scripts to loop and consume environment vars in order to make them docker-friendly. Suggest closing this issue.

sparky8512, 2021-01-12:
I concur, but I'll leave it up to you as you opened it and did the work on it.

closed 2021-01-12 by neurocis