sparky8512/starlink-grpc-tools #2: FR: Send to InfluxDB / MQTT

neurocis, 2020-12-27:
Send the data to an InfluxDB for later consumption by another tool ( Graphana ? )
MQTT for action by IoT systems ( Home Assistant / OpenHAB ? )

sparky8512, 2020-12-27:
Definitely need a better data backend. I'm not particularly expert on databases, so I just threw in the CSV output to start collecting data while figuring that all out.

The first step in this probably needs to be refactoring the scripts so there isn't as much duplication between the JSON parser and the one that does direct gRPC calls. Some duplication is hard to avoid given that the input data winds up structured differently, but the data output should be the exact same between the two.

Equinox-, 2020-12-28:
I wrote this extremely hacky python script based on this that publishes to InfluxDB.  (Prone to failure, so should be wrapped in an auto-restarting system of some kind)

Definitely wouldn't say it's a good solution as is, but it is sufficient for my Grafana needs right now.

```
import grpc, time
from influxdb import InfluxDBClient
from influxdb import SeriesHelper

import spacex.api.device.device_pb2
import spacex.api.device.device_pb2_grpc

class DishyStatusSeries(SeriesHelper):
    class Meta:
        series_name = "spacex.starlink.dishy.status"
        fields = [
            "hardware_version",
            "software_version",
            "state",
            "alert_motors_stuck",
            "alert_thermal_throttle",
            "alert_thermal_shutdown",
            "alert_unexpected_location",
            "snr",
            "seconds_to_first_nonempty_slot",
            "pop_ping_drop_rate",
            "downlink_throughput_bps",
            "uplink_throughput_bps",
            "pop_ping_latency_ms",
            "currently_obstructed",
            "fraction_obstructed"]
        tags = ["id"]

influxClient = InfluxDBClient("host", port, "username", "password", "database", ssl=True, retries=1, timeout=15)
try:
    channel = grpc.insecure_channel("192.168.100.1:9200")

    pending = 0
    count = 0
    while True:
        stub = spacex.api.device.device_pb2_grpc.DeviceStub(channel)
        response = stub.Handle(spacex.api.device.device_pb2.Request(get_status={}))
        dishy = response.dish_get_status
        DishyStatusSeries(
            id=dishy.device_info.id,
            hardware_version=dishy.device_info.hardware_version,
            software_version=dishy.device_info.software_version,
            state=dishy.state,
            alert_motors_stuck=dishy.alerts.motors_stuck,
            alert_thermal_throttle=dishy.alerts.thermal_throttle,
            alert_thermal_shutdown=dishy.alerts.thermal_shutdown,
            alert_unexpected_location=dishy.alerts.unexpected_location,
            snr=dishy.snr,
            seconds_to_first_nonempty_slot=dishy.seconds_to_first_nonempty_slot,
            pop_ping_drop_rate=dishy.pop_ping_drop_rate,
            downlink_throughput_bps=dishy.downlink_throughput_bps,
            uplink_throughput_bps=dishy.uplink_throughput_bps,
            pop_ping_latency_ms=dishy.pop_ping_latency_ms,
            currently_obstructed=dishy.obstruction_stats.currently_obstructed,
            fraction_obstructed=dishy.obstruction_stats.fraction_obstructed)
        pending = pending + 1
        print("Samples: " + str(pending))
        if count > 5:
            try:
                DishyStatusSeries.commit(influxClient)
                print("Wrote " + str(pending))
                pending = 0
            except Exception as e:
                print("Failed to write: " + str(e))
            count = 0
        count = count + 1
        time.sleep(30)
finally:
    # Flush on failure
    DishyStatusSeries.commit(influxClient)
```

sparky8512, 2020-12-28:
> I wrote this extremely hacky python script based on this repo that publishes to InfluxDB. (Prone to failure, so should be wrapped in an auto-restarting system of some kind)

I'm OK with hacky. And the script(s) for get_status are fine to just duplicate wholesale, given how simple that request is.

sparky8512, 2021-01-09:
I forgot to reference this issue in the commit comment, but f067f08952e98d7330a0f71f6f45605e1a2b6b90 adds InfluxDB and MQTT support for the history stats.

Next up is HTTPS/SSL options in those scripts and then retrofitting the command line option handling to the status info scripts.

I also want to implement an option for interval repeat loops to all the scripts, but will probably track that with a separate issue.

sparky8512, 2021-01-11:
I'm going to call this one done. Error printing is a mess and only one of the scripts supports periodic polling, but I have filed separate issues for those.

Let's see, that 1 issue closed and only 2 issued opened as a result. At this rate, I'll be done in (does a little math...) ∞ time!

closed 2021-01-11 by sparky8512