sparky8512/starlink-grpc-tools #18: Docker unable to import trasnceiver_pb2

fudnet, 2021-02-10:
Attempting to use the provided Docker container and output the data to influxdb. Running with these options:

> docker run -d -t --name='starlink-grpc-tools' -e INFLUXDB_HOST=192.168.2.201 -e INFLUXDB_PORT=49153 -e INFLUXDB_DB=starlinkstats neurocis/starlink-grpc-tools dishStatusInflux.py -v

And this is the log output:

> Traceback (most recent call last):
>  File "/app/dishStatusInflux.py", line 23, in <module>
>    import spacex.api.device.device_pb2
>  File "/app/spacex/api/device/device_pb2.py", line 18, in <module>
>    from spacex.api.device import transceiver_pb2 as spacex_dot_api_dot_device_dot_transceiver__pb2
>ImportError: cannot import name 'transceiver_pb2' from 'spacex.api.device' (unknown location)

sparky8512, 2021-02-10:
This is likely the result of the dish software adding one or more new protocol files to the gRPC service. I noticed there were some new request messages listed in SpaceX.API.Device.Request.

The docker container pulls the protocol data dynamically when it starts up and only pulls the files it knows about. While we don't need anything in the new protocol files, the top-level one defining the service still imports them, so will fail if they are missing.

I'm working on getting reflection working directly in the Python scripts, which will eliminate the need for this, but in the mean time, we can just add the new proto files. I'll have a look tomorrow morning and update the instructions in the README and I can update entrypoint.sh so the docker container starts running the new instructions, but @neurocis will need to update the published image.

sparky8512, 2021-02-10:
It looks like the published image has been updated with this change, so this issue should now be resolved. I did also update the README, as this would affect running outside of docker as well, the next time someone ran (or reran) the protoc commands.

closed 2021-02-10 by sparky8512