sparky8512/starlink-grpc-tools #20: seconds_obstructed confusion

dustin, 2021-02-26:
Hi, I'm pushing some data from my starlink into influxdb using your tools.  I'm a grateful data hoarder.

I'm a bit confused by `seconds_obstructed`, though.  The docs suggest it's the number of seconds of downtime in the last 12 hours, but the numbers don't match for me.

I'm currently getting the following:

```
> select last(seconds_obstructed) from "spacex.starlink.user_terminal.status"
name: spacex.starlink.user_terminal.status
time                last
----                ----
1614366741000000000 1933
```

However, the web UI suggests I have much less downtime:

[image]

The mobile app says the same, but 12 hours instead of 24 hours.

It's not clear what's incorrect, but I'd like a happier dashboard.  :)

sparky8512, 2021-02-26:
It is a little confusing, I admit. This is how I currently have `seconds_obstructed` documented (emphasis added):

The amount of time within the history buffer (currently the smaller of 12 hours or since last reboot), in seconds that the user terminal determined to be obstructed, **regardless of whether or not packets were able to be transmitted or received**. ...

I tried to make it clear that `seconds_obstructed` does not necessarily mean downtime. The dish has some algorithm for determining whether or not a sample period is "obstructed" or not that I can only guess involves how often the direction between it and the satellite during that second experienced packet loss (ping drop) in the past. `seconds_obstructed`, as well as `count_obstructed` in the ping_drop history stats group, and the `obstructed` flag in the bulk history data are all raw indications of this determination. They need to be combined with the pop_ping_drop_rate history data to get to the total numbers listed under "Last 12 hours" in the app.

If you're looking to record that data over time into a dashboard, you should look at the `count_full_obstructed_ping_drop`, `count_full_unscheduled_ping_drop`, and `count_full_ping_drop` fields of the ping_drop mode group, which should reflect Obstructed, No Satellites, and total Obstructed+No Satellites+Beta Downtime, respectively. I think you'll probably need to use both --all-samples and  --no-counter command line options, too. I wasn't really thinking of that use case when I made counter tracking the default for ping history stats.

BTW: The "Last 24 hours" label in the web app is misleading. The web app appears to still be an older version. The Android and iOS apps fixed that some time ago to correctly reflect the amount of valid history data, which will be 12 hours or number of hours since reboot if that is less than 12.

dustin, 2021-02-26:
Oooh, that makes a lot of sense.  I did want this number, but not necessarily in a big red box on my UI.

I'm not finding the ping counters, though.  I've turned on all the things I can find:

```docker run --rm -t --name='starlink-grpc-tools' -e INFLUXDB_HOST=X -e INFLUXDB_PORT=8086 -e INFLUXDB_DB=starlink neurocis/starlink-grpc-tools dish_grpc_influx.py -t 10 --all-samples --no-counter -v status obstruction_detail ping_drop ping_run_length ping_latency ping_loaded_latency usage alert_detail```

And that yields the following:

```
> show field keys from "spacex.starlink.user_terminal.status"
name: spacex.starlink.user_terminal.status
fieldKey                          fieldType
--------                          ---------
alert_mast_not_near_vertical      boolean
alert_motors_stuck                boolean
alert_slow_ethernet_speeds        boolean
alert_thermal_shutdown            boolean
alert_thermal_throttle            boolean
alert_unexpected_location         boolean
alerts                            integer
currently_obstructed              boolean
downlink_throughput_bps           float
fraction_obstructed               float
hardware_version                  string
pop_ping_drop_rate                float
pop_ping_latency_ms               float
raw_wedges_fraction_obstructed_0  float
raw_wedges_fraction_obstructed_1  float
raw_wedges_fraction_obstructed_10 float
raw_wedges_fraction_obstructed_11 float
raw_wedges_fraction_obstructed_2  float
raw_wedges_fraction_obstructed_3  float
raw_wedges_fraction_obstructed_4  float
raw_wedges_fraction_obstructed_5  float
raw_wedges_fraction_obstructed_6  float
raw_wedges_fraction_obstructed_7  float
raw_wedges_fraction_obstructed_8  float
raw_wedges_fraction_obstructed_9  float
seconds_obstructed                float
seconds_to_first_nonempty_slot    float
snr                               float
software_version                  string
state                             string
uplink_throughput_bps             float
uptime                            integer
valid_s                           float
wedges_fraction_obstructed_0      float
wedges_fraction_obstructed_1      float
wedges_fraction_obstructed_10     float
wedges_fraction_obstructed_11     float
wedges_fraction_obstructed_2      float
wedges_fraction_obstructed_3      float
wedges_fraction_obstructed_4      float
wedges_fraction_obstructed_5      float
wedges_fraction_obstructed_6      float
wedges_fraction_obstructed_7      float
wedges_fraction_obstructed_8      float
wedges_fraction_obstructed_9      float
```

sparky8512, 2021-02-26:
Those go to a different measurement / series name. Try `spacex.starlink.user_terminal.ping_stats`. I guess I should add that detail to the script module docs.

dustin, 2021-02-26:
Oh yes, I see now.

I also tried out the mqtt thing, which is pretty nice.  I may consider just running that one since I've got a lot of infrastructure around mqtt, though your influxdb app does all the important things (e.g., spooling since my influxdb is on the other side of a vpn).

Thanks for all this work.

sparky8512, 2021-02-26:
Cool. Let me know if you poke at the MQTT support and find it lacking. I mostly just implemented that because it was mentioned as a possibility and I already have a MQTT broker running on my home network (and because it was easy enough), but I don't use it for anything like what these tools spit out.

sparky8512, 2021-02-28:
I just added the note about measurement names, as I mentioned a few comments back. I was going to try to add some clarify text to `starlink_grpc`, too, but I think that would be better served by a Wiki article, probably just expanding a bit on my first comment above. I'll leave this issue open until I get around to that.