sparky8512/starlink-grpc-tools #21: dashboard enhancements

dustin, 2021-02-26:
Is there a good place to discuss dashboards around?  Perhaps Wiki or Discussions?

I've taken the data provided here and made a dashboard that I found more useful for my own needs.  I wanted to share it with others in case anyone is interested.

[image]

Interactive example:  https://snapshot.raintank.io/dashboard/snapshot/8FO8GbRlK4G6hGKtASVQlJ9NnmyDaIVo

Dashboard for import:

[Starlink Statistics.json.zip](https://github.com/sparky8512/starlink-grpc-tools/files/6051951/Starlink.Statistics.json.zip)

sparky8512, 2021-02-26:
I can open up either Wiki or Discussions if you want. Honestly, most interactive discussion to date has been in issue reports, which is probably not ideal. Wiki is probably not great for discussion, either, but might be a good place to showcase what's available, like the screenshot you just posted. I guess Discussion would make more sense for... well... discussion, but I've never actually seen a GitHub project use it. I'm willing to give it a go if people are interested, though. You're not the first person to ask.

dustin, 2021-02-26:
I've only seen Discussions in use in the tesla API thing where there's actually a lot of people trying to work on building APIs for interacting with tesla backends and bug discussions get out of control in such a way that makes the bug status less helpful.

Wiki might be OK.  I don't want to leave junk in bugs that can't be closed or project forks or whatever just to have peripheral contribution.  This is along the lines of "look at stuff that can be done with this project" and not so much the project itself.  The project enables so much great stuff.

sparky8512, 2021-02-26:
OK, I have opened up the Wiki section of this project so that anyone with a GitHub login should be able to add or edit pages. At some point, I'll try to at least add an overview page that doesn't just duplicate the README, but feel free to add pages in the mean time.

dustin, 2021-02-28:
I made https://github.com/sparky8512/starlink-grpc-tools/wiki/Dashboards where I could drop the dashboard I'm using and hopefully bait other people to do interesting things.

closed 2021-02-28 by dustin