sparky8512/starlink-grpc-tools #28: Support for fast_switching and fast_switching_enabled?

NelsonMinar, 2021-06-27:
Apparently back in April with firmware revision `feb4dfde` a new method was enabled called "fast_switching". I took a look and I don't think starlink-grpc-tools supports it. Should it be added? You should be able to see details on it in the `.protoset` I uploaded for another new method: https://github.com/sparky8512/starlink-grpc-tools/files/6640354/dish.zip

The question is what this API call does. I'm thinking / hoping that this is related to a feature they announced in email in April where Dishy will switch to a new satellite not just when the old one is going out of the circular field of view, but also when the satellite is about to be obscured by a site-specific obstruction. That's a major change that should greatly improve reliability for folks like me who have just a couple of trees in view. AFAICT this feature has never actually been turned on. At least I've never noticed a performance improvement (and thank you to your software for letting me monitor that.) Some folks on Reddit think it does work for them but it's hard to tell how reliable that report is.

Anyway, I'm wondering what the method does. From the phrasing I suspect it's a status detection only but maybe it's also a way to toggle the feature off and on?

NelsonMinar, 2021-06-27:
Some discussion of this method and some other new ones at the same time: https://www.reddit.com/r/Starlink/comments/mw0k3v/new_dish_software_comes_with_some_extra_grcp/

sparky8512, 2021-06-27:
I'll have a look when I get to a point where I'm OK with experimenting with my broadband connection, which may not be for a while. I thought I had seen this in the protocol detail last time I looked, tried it, and found it not useful, but it may not have been implemented at that point.

From the protocol description, it does look like an actual setter method, not just status polling. The response has more fields than the request, though, so it looks like probably meant for both. However, when I run it with an empty request, I get no status reported back, so maybe it's still not really implemented, at least in the production firmware. The name of the request implies that it's for EMC testing, which is not something you would need enabled in end-user equipment.

BTW: I'm skeptical that this really relates to the obstruction-avoidance switching, as that has little to do with "fast" and I would not think it would affect EMC in a significant way. FWIW, I'm pretty sure that did get activated on my dish a week or 2 prior to that beta update email, as my obstructed time went way down at that point (and then later went up again and then back down). Unfortunately, my script had stopped running for a few weeks, so I'm missing a chunk of that data.

NelsonMinar, 2021-06-27:
Thanks for all the thoughts! You're right in the context of those other new methods mentioned in the Reddit thread it's not at all clear this is related to obstruction avoidance. 

Interesting there's a set/get method though. I am in a position to experiment with these settings (I have a second ISP).

sparky8512, 2021-06-27:
> I am in a position to experiment with these settings (I have a second ISP).

Can you try it with grpcurl? I believe the syntax would be:
`grpcurl -plaintext -d {\"dish_emc\":{\"fast_switching\":true}} 192.168.100.1:9200 SpaceX.API.Device.Device/Handle`
and can try with the other fields in SpaceX.API.Device.DishEmcRequest, too.

This may be one of those things that's just simpler to do in grpcurl instead of a python script, anyway, similar to the stow and reboot commands.

NelsonMinar, 2021-06-27:
Sure I'll give it a try. I just set it to true using `grpcui` to do the equivalent of your suggestion, I still have Internet access. Will report back in 24 hours. It strikes me "EMC" probably stands for "Electromagnetic Compatibility" and given the various other kinds of radio control stuff in the `dish_emc` API I could believe that this refers to frequency switching or something as much as satellite.

I'm not experienced with gRPC; would you expect `dish_emc` to be both a getter and a setter? Calling it with no arguments (or calling it with `fast_switching` enabled) it returns a block of status data. However it's all set to 0 or False which makes me think it's not returning real data, even simple fields like `timestamp` or `snr`. Even the response to my attempt to set `fast_switching: true` says `fast_switching: false`. I see why you thought this method may not be implemented.

Too bad, it looks like it could have some really interesting info on status of the radio.. `rx_channel`, `voltage`, `theta` and `phi` which could be where the dish is (virtually) pointing... I've attached the response I got in request to my set attempt. 
[response.txt](https://github.com/sparky8512/starlink-grpc-tools/files/6722091/response.txt)

sparky8512, 2021-06-27:
> I'm not experienced with gRPC; would you expect `dish_emc` to be both a getter and a setter?

In general, gRPC method definitions are pretty flexible and support 1 or more structs in as arguments and 1 or more structs out as return, and the structs can be anything supported by protocol buffers.

The `SpaceX.API.Device.Device/Handle` method implements single struct in, single struct out, which is good for basic command or query one-shot calls, but also allows for bi-directional calls where you pass some meaningful data in and get some meaningful data back.

So I guess this comes down to more of a general API question. Normally, I wouldn't dual-purpose a method like that unless changing some of the settings would affect the other ones in some way and you want to get back the full results in one atomic call. But different people approach API design differently. I probably would not have mashed all the requests into a single method, either.

> Calling it with no arguments (or calling it with `fast_switching` set to `true`) it returns a block of status data. However it's all set to 0 or False which makes me think it's not returning real data, even simple fields like `timestamp` or `snr`.

Those are likely the default values per data type. The protocol buffer spec allows for omitting fields whose value is the default value for the data type (usually 0 or equivalent). I guess grpcurl does not fill in the default values.

NelsonMinar, 2021-06-28:
Thanks for the info. After 24 hours my attempt to set `fast_switching` didn't make any measureable change in Starlink performance. Maybe `fast_switching` is already turned on. Or maybe the API isn't implemented at all internally, given it's not returning any data. Although some other unimplemented APIs throw an error...

Still a little confused on the possibility of an overloaded get/set. There's definitely fields in the data structure that it makes little or no sense to set: timestamps, lat/lon coordinates, etc. So that suggests it should work as a get, too. OTOH the other API calls that return info all have `get` in their name. This one's a bit of an oddball.

I'm going to close the ticket; I agree it doesn't make sense to put a lot of work into supporting this method in your code. If it ever starts returning useful info it may be worth revisiting.

closed 2021-06-28 by NelsonMinar