sparky8512/starlink-grpc-tools #38: Docker image doesn't contain dish_grpc_influx2.py

StephenShamakian, 2022-01-03:
I am moving over to InfluxDB v2 and ran into an issue.

I get this error when using the new docker image located here: ghcr.io/sparky8512/starlink-grpc-tools

/usr/local/bin/python3: can't open file '/app/dish_grpc_influx2.py': [Errno 2] No such file or directory

It looks like the docker image was last updated 2 months ago. Yet the influx v2 code was added after the last docker image update.

sparky8512, 2022-01-03:
Yeah. Sorry, I was going to add a few more changes before spinning a new Docker image, but never got around to it. I'll kick off a Docker image build right now.

sparky8512, 2022-01-03:
OK, it's updated now and "latest" tag has been advanced to point to the new image.

Note that all the issues I commented on in pull request #37 are still present. I think the only significant ones are that some of the command line options for SSL are broken: if your server requires SSL, use an https URL (--secure option does nothing) and the --ca-cert option is broken, but if I recall correctly, the --insecure option still works for the case where a self-signed cert (or similar) is needed.

closed 2022-01-03 by sparky8512

StephenShamakian, 2022-01-03:
@sparky8512 No SSL here for InfluxDB. It's all local docker network connections for my setup. I have been seeing a lot of issues with https and InfluxDB v2.0. Another device I have that imports data into my InfluxDB also has mentions of https troubles as well with v2.x.

Anyway thanks! Working now!