sparky8512/starlink-grpc-tools #42: obstruction_interval=nan causes influx ingest to fail

linuxkidd, 2022-03-24:
Using `dish_grpc_influx.py`, noticed that my statistics stopped being ingested.  Checked logs and found many gigabytes of the following:
```
unable to parse 'spacex.starlink.user_terminal.status,id=ut000#####-########-######## alert_mast_not_near_vertical=False,alert_motors_stuck=False,alert_roaming=False,alert_slow_ethernet_speeds=False,alert_thermal_shutdown=False,alert_thermal_throttle=False,alert_unexpected_location=False,alerts=0i,currently_obstructed=False,direction_azimuth=-1.9601364135742188,direction_elevation=63.73470687866211,downlink_throughput_bps=7314.8662109375,fraction_obstructed=0.00020500205573625863,hardware_version=\"rev1_pre_production\",obstruction_duration=0.0,obstruction_interval=nan,pop_ping_drop_rate=0.0,pop_ping_latency_ms=31.85714340209961,raw_wedges_fraction_obstructed_0=0.0,raw_wedges_fraction_obstructed_1=0.0,raw_wedges_fraction_obstructed_10=0.0,raw_wedges_fraction_obstructed_11=0.0,raw_wedges_fraction_obstructed_2=0.0,raw_wedges_fraction_obstructed_3=0.0,raw_wedges_fraction_obstructed_4=0.0,raw_wedges_fraction_obstructed_5=0.0,raw_wedges_fraction_obstructed_6=0.0,raw_wedges_fraction_obstructed_7=0.0,raw_wedges_fraction_obstructed_8=0.0,raw_wedges_fraction_obstructed_9=0.0,seconds_to_first_nonempty_slot=0.0,software_version=\"ffbba606-958e-40c1-9668-b8f1cbf13081.uterm.release\",state=\"CONNECTED\",uplink_throughput_bps=10368.3193359375,uptime=270355i,valid_s=269986.0,wedges_fraction_obstructed_0=0.0,wedges_fraction_obstructed_1=0.0,wedges_fraction_obstructed_10=0.0,wedges_fraction_obstructed_11=0.0,wedges_fraction_obstructed_2=0.0,wedges_fraction_obstructed_3=0.0,wedges_fraction_obstructed_4=0.0,wedges_fraction_obstructed_5=0.0,wedges_fraction_obstructed_6=0.0,wedges_fraction_obstructed_7=0.0,wedges_fraction_obstructed_8=0.0,wedges_fraction_obstructed_9=0.0 1648095497': invalid number dropped=0"
```

I tried modifying `starlink_grpc.py` line 622 as follows:
```
    obstruction_interval = float(status.obstruction_stats.avg_prolonged_obstruction_interval_s or 0)
```
Alas, this didn't seem to help.

sparky8512, 2022-03-24:
The change in 9ca1a8a51c62dd522ec7b6db29e75e0f23a14ab8 would have exposed this.

Looks like the dish reports the obstruction interval as NaN if it experiences absolutely no obstructions, in which case, the reporting of these data items should probably just be skipped, which is what was always happening before 9ca1a8a51c62dd522ec7b6db29e75e0f23a14ab8. Maybe this is what that valid flag was originally supposed to be protecting, but then they never actually implemented it that way, but instead reported the no obstructions case as (0, NaN).

linuxkidd, 2022-03-24:
Ah, I was assuming the `nan` was a side effect of the conversions from `None` type in Python.   It's interesting though that the `float(val or 0)` didn't correct it though, since the string `nan` would clearly not convert to float...  but it still shows exactly the same way in the output.

closed 2022-03-26 by sparky8512

sparky8512, 2022-03-26:
@linuxkidd: Could you please pick up that change and test on your installation? I was able to reproduce the reporting of NaN values, as it happens for a few minutes after dish reboot even if there are occasional obstructions, but I don't get them all the time the way you apparently do.

With change 35150803efe6e18460ad517e7741134302ac5a5e, the lower layer goes back to reporting this data as (None, None), same as was happening prior to change 9ca1a8a51c62dd522ec7b6db29e75e0f23a14ab8. I believe the Influx client handles that by just not including those fields in the data point.

sparky8512, 2022-03-26:
BTW: This NaN value behavior also came up in the [review comments on pull request #40](https://github.com/sparky8512/starlink-grpc-tools/pull/40/commits/323fc26754a62fe005b233b80a2eabf10146b779#r818118032). I probably should have paid more attention to it then, but I though it was more of a corner case than it apparently is.

linuxkidd, 2022-03-27:
Hi @sparky8512  -- Looks good!  no more 'nan' errors and my data is making its way into influx again.  Thanks!

commit 35150803 (2022-03-26): New validity check for prolonged obstruction data

Reintroduce validity check for prolonged obstruction duration and interval, but based on checking for 0.0 and NaN values instead of relying on the valid flag in the gRPC message. Empirical evidence suggests that the dish reports (0, NaN) for those 2 values when it has no obstruction data with which to work, such as when there actually are no obstructions or when the dish has recently rebooted.

This should fix #42.