sparky8512/starlink-grpc-tools #43: dish_unstow command

itemir, 2022-04-09:
Thank you for this repo.

Using `grpcurl`, it is easy to invoke the `dish_stow` command but I could not find the corresponding command to unstow it. Tried the `dish_stow` (toggling), `dish_unstow`, `unstow` to no avail. Any idea what it is?

sparky8512, 2022-04-09:
I don't think there is an explicit `unstow`. Back when the public beta first started, I think you had to actually power cycle to get out of stow mode, but these days, I think a `reboot` will do it.

itemir, 2022-04-09:
There is now an explicit unstow option on the app and the web page. And it just unstows the dish as expected. I tried the `reboot` gRPC command. It did work but unsurprisingly it did reboot the system. When you unstow it from the app, the behavior is significantly different so I suspect there is a command but not sure what it is. I found a bunch of gRPC commands in the bundled JS and some code to unstow the dish but could not decipher it so far:

    REQUEST_NOT_SET:0
    SIGNED_REQUEST:15
    GET_NEXT_ID:1006
    AUTHENTICATE:1005
    ENABLE_FLOW:1018
    FACTORY_RESET:1011
    GET_DEVICE_INFO:1008
    GET_HISTORY:1007
    GET_LOG:1012
    GET_NETWORK_INTERFACES:1015
    GET_PING:1009
    PING_HOST:1016
    GET_STATUS:1004
    REBOOT:1001
    SET_SKU:1013
    SET_TRUSTED_KEYS:1010
    SPEED_TEST:1003
    UPDATE:1014
    GET_LOCATION:1017
    GET_HEAP_DUMP:1019
    RESTART_CONTROL:1020
    FUSE:1021
    GET_PERSISTENT_STATS:1022
    GET_CONNECTIONS:1023
    FLUSH_TELEM:1026
    START_SPEEDTEST:1027
    GET_SPEEDTEST_STATUS:1028
    REPORT_CLIENT_SPEEDTEST:1029
    INITIATE_REMOTE_SSH:1030
    SELF_TEST:1031
    SET_TEST_MODE:1032
    DISH_STOW:2002
    DISH_GET_CONTEXT:2003
    DISH_GET_OBSTRUCTION_MAP:2008
    DISH_SET_EMC:2007
    DISH_GET_EMC:2009
    DISH_SET_CONFIG:2010
    DISH_GET_CONFIG:2011
    TRANSCEIVER_IF_LOOPBACK_TEST:4001
    TRANSCEIVER_GET_STATUS:4003
    TRANSCEIVER_GET_TELEMETRY:4004
    WIFI_GET_CLIENTS:3002
    WIFI_GET_DIAGNOSTICS:3008
    WIFI_GET_PING_METRICS:3007
    WIFI_SET_CONFIG:3001
    WIFI_GET_CONFIG:3009
    WIFI_SETUP:3003
    WIFI_SET_MESH_DEVICE_TRUST:3012
    WIFI_SET_MESH_CONFIG:3013
    WIFI_GET_CLIENT_HISTORY:3015

sparky8512, 2022-04-10:
OK, I can trace what command the app is using for that, then.

sparky8512, 2022-04-10:
Ah, I see. It is using the `dish_stow` request message, but that message has an `unstow` field. Since it's a bool field, it will default to false, so an empty `dish_stow` will stow (unstow=false). To unstow, you need to pass a request that has that set to true. With `grpcurl`, that would be something like:
```
grpcurl -plaintext -d {\"dish_stow\":{\"unstow\":true}} 192.168.100.1:9200 SpaceX.API.Device.Device/Handle
```
(adjust quoting as necessary)

This is probably worth documenting somewhere, if not a (small) dedicated script to do it.

itemir, 2022-04-10:
Awesome, thanks! Happy to add a shell script to stow/unstow if you'd like.

Out of curiosity, how did you find this key? I looked around but it didn't jump at me.

sparky8512, 2022-04-10:
I usually just use `grpcurl` `describe` mode on the main request message and then look at any of the specific requests that look new or that I know change from time to time, but since this one snuck in some time ago without me noticing, I figured it would be easier to run a packet capture with Wireshark. That only gives you the binary protobuf data, so I then fed that through an online protobuf decoder and then looked up the message and field type IDs in the `grpcurl` `describe` output.

sparky8512, 2022-04-19:
In addition to the new script, I also add [a wiki page](https://github.com/sparky8512/starlink-grpc-tools/wiki/Useful-grpcurl-commands) with the grpcurl commands for reboot, stow, and unstow.

That should do it for this issue.

closed 2022-04-19 by sparky8512