sparky8512/starlink-grpc-tools #44: buffering when stdout is not a terminal

joeyh, 2022-04-12:
python3 dish_grpc_text.py status -t 1|cat

I would expect this to display one line per second, but it seems to buffer a lot of lines, and then display them all.

My use case is parsing the status stream with a script to get obstruction information into my desktop's panel..

joeyh, 2022-04-12:
I am not a python developer, but I read https://stackoverflow.com/questions/107705/disable-output-buffering and found this patch works:
```

diff --git a/dish_grpc_text.py b/dish_grpc_text.py
index 29b844e..52821e9 100644
--- a/dish_grpc_text.py
+++ b/dish_grpc_text.py
@@ -110,7 +110,7 @@ def parse_args():
 def open_out_file(opts, mode):
     if opts.out_file == "-":
         # open new file, so it can be closed later without affecting sys.stdout
-        return os.fdopen(sys.stdout.fileno(), "w", closefd=False)
+        return os.fdopen(sys.stdout.fileno(), "w", 1, closefd=False)
     return open(opts.out_file, mode, buffering=1)
```

sparky8512, 2022-04-12:
By default, Python will buffer stdout unless it is a TTY, so redirecting into a pipe (or a file, for that matter) will cause it to buffer.

I thought about forcing it to line buffer, at least in the text output script, but that seems to me that it would be taking away flexibility in case someone actually wants it to buffer. I also thought about making it a command line option, but there is already a command line option to the `python` executable that will prevent it from buffering: the `-u` option. For example:
```
python3 -u dish_grpc_text.py status -t 1|cat
```
should do what you want.

I could probably be convinced that it would be better to just always line buffer in that script. In my own usage, I use that `-u` interpreter option because I redirect to a file and I try to minimize data loss when the power goes out, and I can't really think of a use case where someone needs it to be block buffered.

joeyh, 2022-04-12:
-u does not work for me. (Python 3.10.4)

Note that this also breaks basic use with grep if you just want to watch a single value for changes while looping..

sparky8512, 2022-04-13:
> -u does not work for me. (Python 3.10.4)

Oh. Yeah. I probably broke this when I changed it to reopen the output file in order to support specifying a file for output.  Even if that weren't the case, I think I agree with you that there really isn't any benefit to block buffering for this script, given the data rate.

I will change it to always line buffer the output.

closed 2022-04-13 by sparky8512