sparky8512/starlink-grpc-tools #46: Doesn't like 3.10.3 or 3.10.4 but is fine with 3.8.13

Crasoum, 2022-06-03:
I get: 
  File "C:\Users\craso\Documents\GitHub\starlink-grpc-tools\dish_grpc_text.py", line 21, in <module>
    import dish_common
  File "C:\Users\craso\Documents\GitHub\starlink-grpc-tools\dish_common.py", line 22, in <module>
    import starlink_grpc
  File "C:\Users\craso\Documents\GitHub\starlink-grpc-tools\starlink_grpc.py", line 351, in <module>
    from spacex.api.device import device_pb2
ModuleNotFoundError: No module named 'spacex'

With Python 3.10.x, running py -3.8 works though.

sparky8512, 2022-06-03:
Works for me using 3.10.4 (on Windows-64).

The error you're getting looks like the `yagrc` package is not installed (or is not working for some reason), so it's looking for manually generated protocol files.

Crasoum, 2022-06-03:
Huh. Odd. I manually pip installed yagrc and it installed protobuf-4.21.1-cp310-abi3-win_amd64.whl and now it works. My bad. I did install the requirements file first before all this and just figured it was a versioning issue. 

Good looking out.

sparky8512, 2022-06-03:
That is weird. Running pip against requirements.txt should have installed everything needed.

There is actually a stated version conflict between grpcio-tools and protobuf 4.x, so pip should have installed an earlier version of protobuf (probably 3.20.1). grpcio-tools is not actually used unless you are manually generating the protocol modules, though, and even then, as far as I can tell, it works just fine with protobuf 4.21. I suspect they just haven't gotten around to updating the version requirements for the grpcio-tools package.

Crasoum, 2022-06-03:
Thanks again!

closed 2022-06-03 by Crasoum

netllama, 2022-08-16:
I'm seeing the same problem, except I'm on Linux (Fedora36) with python-3.10.5:

```
$ pip install --upgrade -r requirements.txt
Collecting grpcio>=1.12.0
  Downloading grpcio-1.47.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (4.5 MB)
     |████████████████████████████████| 4.5 MB 1.8 MB/s 
Collecting grpcio-tools>=1.20.0
  Downloading grpcio_tools-1.47.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (2.4 MB)
     |████████████████████████████████| 2.4 MB 2.0 MB/s 
Collecting protobuf>=3.6.0
  Downloading protobuf-4.21.5-cp37-abi3-manylinux2014_x86_64.whl (408 kB)
     |████████████████████████████████| 408 kB 1.9 MB/s 
Collecting yagrc>=1.1.1
  Downloading yagrc-1.1.1-py3-none-any.whl (15 kB)
Collecting paho-mqtt>=1.5.1
  Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
     |████████████████████████████████| 99 kB 1.6 MB/s 
Collecting influxdb>=5.3.1
  Downloading influxdb-5.3.1-py2.py3-none-any.whl (77 kB)
     |████████████████████████████████| 77 kB 2.0 MB/s 
Collecting influxdb_client>=1.23.0
  Downloading influxdb_client-1.31.0-py3-none-any.whl (705 kB)
     |████████████████████████████████| 705 kB 1.6 MB/s 
Collecting pypng>=0.0.20
  Downloading pypng-0.20220715.0-py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 1.8 MB/s 
Collecting six>=1.5.2
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting protobuf>=3.6.0
  Downloading protobuf-3.20.1-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 1.4 MB/s 
Requirement already satisfied: setuptools in /home/netllama/stuff/sl/sl/lib/python3.10/site-packages (from grpcio-tools>=1.20.0->-r requirements.txt (line 2)) (57.4.0)
Collecting grpcio-reflection>=1.7.3
  Downloading grpcio_reflection-1.47.0-py3-none-any.whl (16 kB)
Collecting requests>=2.17.0
  Downloading requests-2.28.1-py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 555 kB/s 
Collecting msgpack
  Downloading msgpack-1.0.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (316 kB)
     |████████████████████████████████| 316 kB 689 kB/s 
Collecting pytz
  Downloading pytz-2022.2.1-py2.py3-none-any.whl (500 kB)
     |████████████████████████████████| 500 kB 890 kB/s 
Collecting python-dateutil>=2.6.0
  Downloading python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
     |████████████████████████████████| 247 kB 368 kB/s 
Collecting certifi>=14.05.14
  Downloading certifi-2022.6.15-py3-none-any.whl (160 kB)
     |████████████████████████████████| 160 kB 341 kB/s 
Collecting rx>=3.0.1
  Downloading Rx-3.2.0-py3-none-any.whl (199 kB)
     |████████████████████████████████| 199 kB 379 kB/s 
Collecting urllib3>=1.26.0
  Downloading urllib3-1.26.11-py2.py3-none-any.whl (139 kB)
     |████████████████████████████████| 139 kB 236 kB/s 
Collecting charset-normalizer<3,>=2
  Downloading charset_normalizer-2.1.0-py3-none-any.whl (39 kB)
Collecting idna<4,>=2.5
  Downloading idna-3.3-py3-none-any.whl (61 kB)
     |████████████████████████████████| 61 kB 265 kB/s 
Using legacy 'setup.py install' for paho-mqtt, since package 'wheel' is not installed.
Installing collected packages: six, urllib3, protobuf, idna, grpcio, charset-normalizer, certifi, rx, requests, pytz, python-dateutil, msgpack, grpcio-reflection, yagrc, pypng, paho-mqtt, influxdb-client, influxdb, grpcio-tools
    Running setup.py install for paho-mqtt ... done
Successfully installed certifi-2022.6.15 charset-normalizer-2.1.0 grpcio-1.47.0 grpcio-reflection-1.47.0 grpcio-tools-1.47.0 idna-3.3 influxdb-5.3.1 influxdb-client-1.31.0 msgpack-1.0.4 paho-mqtt-1.6.1 protobuf-3.20.1 pypng-0.20220715.0 python-dateutil-2.8.2 pytz-2022.2.1 requests-2.28.1 rx-3.2.0 six-1.16.0 urllib3-1.26.11 yagrc-1.1.1
```