sparky8512/starlink-grpc-tools #48: Is it possible to to modify the setting of the snow melt configuration?

Llaves, 2022-06-13:
Is this reachable by the API?

sparky8512, 2022-06-14:
It is, but I get a permission denied error when I try to do it via grpcurl.

I can look into how the mobile app is doing that, but it looks like it requires logging in to your Starlink account, so it may be directing that though a remote connection and/or invoking an authentication involving SpaceX servers.

Llaves, 2022-06-14:
As long as the app still lets me do it through the remote access (I run in bypass mode), it's not that important to me.

closed 2022-06-14 by Llaves

sparky8512, 2022-06-14:
For the record: I verified via a packet capture that the mobile app does not change this setting directly via grpc request to the dish. Rather, it appears to be communicating via HTTPS to a server (api2.starlink.com, I think, although it talks to several), which presumably pokes the dish remotely.

It _is_ possible to view the current configuration using grpc, but that currently only includes this setting, and it's just going to be whatever it was last set to.