sparky8512/starlink-grpc-tools #49: Status works but write to influx returns "No host specified"

SkyMoCo, 2022-07-06:
python3 dish_grpc_text.py -s 2 status works returns correctly with current data, but dish_grpc_influx2 returns an error.   

I don't see any option to specify a host and the url works just perfect.

 python3 dish_grpc_influx2.py -v -u "localhost:8086" -T "<token>" -B StarLink -O BVRanch -s 2 status
Data points queued: 1
Data points queued: 1
Data points written: 1
ERROR: The batch item wasn't processed successfully because: No host specified.

SkyMoCo, 2022-07-06:
What about also allowing at config file?     gstate.influx_client = InfluxDBClient.from_config_file("influx2.ini")

Great program BTW...

sparky8512, 2022-07-06:
> What about also allowing at config file? gstate.influx_client = InfluxDBClient.from_config_file("influx2.ini")

If you just want to read options from a file, you can do so with an @FILNAME arg (see very end of help text). That may not be as pretty as a config file, but it works across all the scripts.

I could still be convinced to add it if you really want it, though, as not all config file options are covered by the command line options.

closed 2022-07-06 by sparky8512