sparky8512/starlink-grpc-tools #53: install problems with python-3.10.x

netllama, 2022-08-16:
On a Fedora35-x86_64 system in a freshly created virtenv, with python-3.10.5, the `spacex` pkg is never installed by default, and even after manually installing it, other packages are missing/broken:

```
(sl) [netllama@hal starlink-grpc-tools]$ pip install --upgrade -r requirements.txt
Collecting grpcio>=1.12.0
  Downloading grpcio-1.47.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (4.5 MB)
     |████████████████████████████████| 4.5 MB 1.8 MB/s 
Collecting grpcio-tools>=1.20.0
  Downloading grpcio_tools-1.47.0-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (2.4 MB)
     |████████████████████████████████| 2.4 MB 2.0 MB/s 
Collecting protobuf>=3.6.0
  Downloading protobuf-4.21.5-cp37-abi3-manylinux2014_x86_64.whl (408 kB)
     |████████████████████████████████| 408 kB 1.9 MB/s 
Collecting yagrc>=1.1.1
  Downloading yagrc-1.1.1-py3-none-any.whl (15 kB)
Collecting paho-mqtt>=1.5.1
  Downloading paho-mqtt-1.6.1.tar.gz (99 kB)
     |████████████████████████████████| 99 kB 1.6 MB/s 
Collecting influxdb>=5.3.1
  Downloading influxdb-5.3.1-py2.py3-none-any.whl (77 kB)
     |████████████████████████████████| 77 kB 2.0 MB/s 
Collecting influxdb_client>=1.23.0
  Downloading influxdb_client-1.31.0-py3-none-any.whl (705 kB)
     |████████████████████████████████| 705 kB 1.6 MB/s 
Collecting pypng>=0.0.20
  Downloading pypng-0.20220715.0-py3-none-any.whl (58 kB)
     |████████████████████████████████| 58 kB 1.8 MB/s 
Collecting six>=1.5.2
  Downloading six-1.16.0-py2.py3-none-any.whl (11 kB)
Collecting protobuf>=3.6.0
  Downloading protobuf-3.20.1-cp310-cp310-manylinux_2_12_x86_64.manylinux2010_x86_64.whl (1.1 MB)
     |████████████████████████████████| 1.1 MB 1.4 MB/s 
Requirement already satisfied: setuptools in /home/netllama/stuff/sl/sl/lib/python3.10/site-packages (from grpcio-tools>=1.20.0->-r requirements.txt (line 2)) (57.4.0)
Collecting grpcio-reflection>=1.7.3
  Downloading grpcio_reflection-1.47.0-py3-none-any.whl (16 kB)
Collecting requests>=2.17.0
  Downloading requests-2.28.1-py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 555 kB/s 
Collecting msgpack
  Downloading msgpack-1.0.4-cp310-cp310-manylinux_2_17_x86_64.manylinux2014_x86_64.whl (316 kB)
     |████████████████████████████████| 316 kB 689 kB/s 
Collecting pytz
  Downloading pytz-2022.2.1-py2.py3-none-any.whl (500 kB)
     |████████████████████████████████| 500 kB 890 kB/s 
Collecting python-dateutil>=2.6.0
  Downloading python_dateutil-2.8.2-py2.py3-none-any.whl (247 kB)
     |████████████████████████████████| 247 kB 368 kB/s 
Collecting certifi>=14.05.14
  Downloading certifi-2022.6.15-py3-none-any.whl (160 kB)
     |████████████████████████████████| 160 kB 341 kB/s 
Collecting rx>=3.0.1
  Downloading Rx-3.2.0-py3-none-any.whl (199 kB)
     |████████████████████████████████| 199 kB 379 kB/s 
Collecting urllib3>=1.26.0
  Downloading urllib3-1.26.11-py2.py3-none-any.whl (139 kB)
     |████████████████████████████████| 139 kB 236 kB/s 
Collecting charset-normalizer<3,>=2
  Downloading charset_normalizer-2.1.0-py3-none-any.whl (39 kB)
Collecting idna<4,>=2.5
  Downloading idna-3.3-py3-none-any.whl (61 kB)
     |████████████████████████████████| 61 kB 265 kB/s 
Using legacy 'setup.py install' for paho-mqtt, since package 'wheel' is not installed.
Installing collected packages: six, urllib3, protobuf, idna, grpcio, charset-normalizer, certifi, rx, requests, pytz, python-dateutil, msgpack, grpcio-reflection, yagrc, pypng, paho-mqtt, influxdb-client, influxdb, grpcio-tools
    Running setup.py install for paho-mqtt ... done
Successfully installed certifi-2022.6.15 charset-normalizer-2.1.0 grpcio-1.47.0 grpcio-reflection-1.47.0 grpcio-tools-1.47.0 idna-3.3 influxdb-5.3.1 influxdb-client-1.31.0 msgpack-1.0.4 paho-mqtt-1.6.1 protobuf-3.20.1 pypng-0.20220715.0 python-dateutil-2.8.2 pytz-2022.2.1 requests-2.28.1 rx-3.2.0 six-1.16.0 urllib3-1.26.11 yagrc-1.1.1
WARNING: You are using pip version 21.2.3; however, version 22.2.2 is available.
You should consider upgrading via the '/home/netllama/stuff/sl/sl/bin/python3 -m pip install --upgrade pip' command.
(sl) [netllama@hal starlink-grpc-tools]$ python3 dump_dish_status.py
Traceback (most recent call last):
  File "/home/netllama/stuff/starlink-grpc-tools/dump_dish_status.py", line 6, in <module>
    from spacex.api.device import device_pb2
ModuleNotFoundError: No module named 'spacex'
(sl) [netllama@hal starlink-grpc-tools]$ pip install spacex
Collecting spacex
  Downloading spacex-0.0.2.tar.gz (5.4 kB)
Collecting requests==2.18.4
  Downloading requests-2.18.4-py2.py3-none-any.whl (88 kB)
     |████████████████████████████████| 88 kB 812 kB/s 
Collecting chardet<3.1.0,>=3.0.2
  Using cached chardet-3.0.4-py2.py3-none-any.whl (133 kB)
Collecting urllib3<1.23,>=1.21.1
  Downloading urllib3-1.22-py2.py3-none-any.whl (132 kB)
     |████████████████████████████████| 132 kB 1.0 MB/s 
Collecting idna<2.7,>=2.5
  Downloading idna-2.6-py2.py3-none-any.whl (56 kB)
     |████████████████████████████████| 56 kB 1.2 MB/s 
Requirement already satisfied: certifi>=2017.4.17 in /home/netllama/stuff/sl/sl/lib/python3.10/site-packages (from requests==2.18.4->spacex) (2022.6.15)
Using legacy 'setup.py install' for spacex, since package 'wheel' is not installed.
Installing collected packages: urllib3, idna, chardet, requests, spacex
  Attempting uninstall: urllib3
    Found existing installation: urllib3 1.26.11
    Uninstalling urllib3-1.26.11:
      Successfully uninstalled urllib3-1.26.11
  Attempting uninstall: idna
    Found existing installation: idna 3.3
    Uninstalling idna-3.3:
      Successfully uninstalled idna-3.3
  Attempting uninstall: requests
    Found existing installation: requests 2.28.1
    Uninstalling requests-2.28.1:
      Successfully uninstalled requests-2.28.1
    Running setup.py install for spacex ... done
ERROR: pip's dependency resolver does not currently take into account all the packages that are installed. This behaviour is the source of the following dependency conflicts.
influxdb-client 1.31.0 requires urllib3>=1.26.0, but you have urllib3 1.22 which is incompatible.
Successfully installed chardet-3.0.4 idna-2.6 requests-2.18.4 spacex-0.0.2 urllib3-1.22
WARNING: You are using pip version 21.2.3; however, version 22.2.2 is available.
You should consider upgrading via the '/home/netllama/stuff/sl/sl/bin/python3 -m pip install --upgrade pip' command.
(sl) [netllama@hal starlink-grpc-tools]$ python3 dump_dish_status.py
Traceback (most recent call last):
  File "/home/netllama/stuff/starlink-grpc-tools/dump_dish_status.py", line 6, in <module>
    from spacex.api.device import device_pb2
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/spacex/__init__.py", line 13, in <module>
    from .launchpad import *
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/spacex/launchpad.py", line 1, in <module>
    from .http import request
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/spacex/http.py", line 1, in <module>
    import requests
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/requests/__init__.py", line 43, in <module>
    import urllib3
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/urllib3/__init__.py", line 8, in <module>
    from .connectionpool import (
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/urllib3/connectionpool.py", line 29, in <module>
    from .connection import (
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/urllib3/connection.py", line 39, in <module>
    from .util.ssl_ import (
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/urllib3/util/__init__.py", line 3, in <module>
    from .connection import is_connection_dropped
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/urllib3/util/connection.py", line 3, in <module>
    from .wait import wait_for_read
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/urllib3/util/wait.py", line 1, in <module>
    from .selectors import (
  File "/home/netllama/stuff/sl/sl/lib64/python3.10/site-packages/urllib3/util/selectors.py", line 14, in <module>
    from collections import namedtuple, Mapping
ImportError: cannot import name 'Mapping' from 'collections' (/usr/lib64/python3.10/collections/__init__.py)
(sl) [netllama@hal starlink-grpc-tools]$ which python
~/stuff/sl/sl/bin/python
(sl) [netllama@hal starlink-grpc-tools]$ python -V
Python 3.10.5
```

sparky8512, 2022-08-16:
The `spacex.api.device` package  does not come from an installable project, those are the generated protocol modules. As noted in the [README](https://github.com/sparky8512/starlink-grpc-tools#other-scripts):

> `dump_dish_status.py` is a simple example of how to use the grpc modules (the ones generated by protoc, not `starlink_grpc`) directly. This script does require the [generated gRPC protocol modules](https://github.com/sparky8512/starlink-grpc-tools/wiki/gRPC-Protocol-Modules), contrary to the above recommendation against generating them.

The other scripts can get the protocol modules via reflection, but `dump_dish_status.py` is intended to be as simple as possible, so needs a bit more preparation.

BTW: I would recommend uninstalling the `spacex` pip package. I don't know what that is, but I'm pretty sure it's not published by SpaceX or related to communicating with a Starlink dish, and it has a Python package namespace conflict with packages that SpaceX uses for the grpc protocols.

netllama, 2022-08-16:
ok, thanks, that clears things up.

closed 2022-08-16 by netllama