sparky8512/starlink-grpc-tools #54: MQTT Mode?

Pacers31Colts18, 2022-08-17:
Having trouble getting this to work with MQTT. Am able to run, but when I try to use the MQTT script, I'm running into issues with the mode option, assuming I'm not doing something right.

Here is what I have

python3 dish_grpc_mqtt.py --hostname xxxxxx --port 1883 --username mqtt-user --password xxxxxxxxx mode usage

I've tried different variations, with 'usage' mode=usage --mode, etc, etc. Any thougts?

sparky8512, 2022-08-17:
The mode argument is positional, so you don't actually include the word 'mode' on the command line, you just list the data group modes you want to emit:
```shell
python3 dish_grpc_mqtt.py --hostname xxxxxx --port 1883 --username mqtt-user --password xxxxxxxxx usage
```

There has to be at least one, though, which is why the usage line in the help output has the first 'mode' outside the braces. That's just how the `argparse` library depicts things.

sparky8512, 2022-08-26:
I'm going to assume this has addressed the problem you were having and close this issue out. If not, feel free to reopen and attach full error output.

closed 2022-08-26 by sparky8512