sparky8512/starlink-grpc-tools #63: TypedDicts have type Any in IDE

boswelja, 2022-09-11:
Example: `(variable) ObstructionDict: Type[ObstructionDict] | Any`
This means `status_data` returns `Any` (except for AlertDict, which appears correct).

I'm playing around locally now, I converted `StatusDict` to a class def like so
```kotlin
class StatusDict(TypedDict):
    id: str
    hardware_version: str
    software_version: str
    state: str
    uptime: int
    snr: Optional[float]
    seconds_to_first_nonempty_slot: float
    pop_ping_drop_rate: float
    downlink_throughput_bps: float
    uplink_throughput_bps: float
    pop_ping_latency_ms: float
    alerts: int
    fraction_obstructed: float
    currently_obstructed: bool
    seconds_obstructed: Optional[float]
    obstruction_duration: Optional[float]
    obstruction_interval: Optional[float]
    direction_azimuth: float
    direction_elevation: float
    is_snr_above_noise_floor: bool
```
Which fixes the `Any` issue, but now I don't get type hints for variables in the class, I suspect this is caused by the `def TypedDict`

boswelja, 2022-09-11:
I just discovered this package that enabled TypedDict: https://pypi.org/project/typing-extensions/
It appears to work correctly locally, including via the existing (not defining classes) way of defining TypedDicts. If you're happy with adding one dependency for this I'll go ahead and open a PR

sparky8512, 2022-09-11:
So this would replace the conditional fallback for Python 3.7? If so, that's fine.

Note that there are 3 different places where the dependencies are listed:
- requirements.txt
- packaging/setup.cfg
- Dockerfile

sparky8512, 2022-09-11:
BTW: While class notation will work for `StatusDict`, it won't for some of the others because the dict keys are not all valid Python syntax, as they have brackets and/or commas.

boswelja, 2022-09-11:
Square braces don't appear to affect it locally, `PingLatencyDict["deciles_all_ping_latency[]"]` appears to work fine. So does `PingDropRlDict["run_minutes[1,]"]` :)

sparky8512, 2022-09-11:
That was poorly worded on my part. My point was not that they are invalid dict keys, it is that they are not valid identifiers, so class notation would break:
```python
class PingLatencyDict(TypedDict):
    deciles_all_ping_latency[]: Sequence[float]
                             ^
SyntaxError: invalid syntax
```
which is one of the reasons I kept the alternative syntax for the `TypedDict` types. Anyway, this is beside the point of this issue.

... and I just realized that `deciles_all_ping_latency` (and a few others) should be marked optional, too. I will fix that later.

closed 2022-09-14 by boswelja