sparky8512/starlink-grpc-tools #65: no attribute 'wedge_abs_fraction_obstructed'

archerne, 2022-11-08:
Hello, I am on starlink version `ee39beae-c399-4648-a7d6-949193d8c910.uterm.release` and release v1.1.0 of the python library. I am currently getting this error:

```  
File "/usr/local/lib/python3.10/site-packages/starlink_grpc.py", line 765, in status_data
    "wedges_fraction_obstructed[]": status.obstruction_stats.wedge_abs_fraction_obstructed,
AttributeError: 'DishObstructionStats' object has no attribute 'wedge_abs_fraction_obstructed'
```

sparky8512, 2022-11-08:
I'm not surprised this happened, although it was a little quicker than I expected. They stopped populating that field a couple firmware releases ago, and now it appears they have dropped it from the protocol.

I don't have that firmware version yet, but it's an easy enough fix.

sparky8512, 2022-11-09:
That change should fix this crash, but I have not verified against the firmware version that trips it up, as I don't have it, and probably won't for at least a few days.

I would prefer not to tag a new release until someone (not necessarily me) verifies it against this newer firmware.

archerne, 2022-11-09:
I can test this in the morning. Approx 10hrs from now.

archerne, 2022-11-09:
This does fix the issue. Getting the status data now results in the correct data being returned and not an error.

sparky8512, 2022-11-09:
Thanks. Fix has now been published as v1.1.1, both to the pip-installable core package and the docker image.

I will put some thought into how to prevent this from causing breaking bugs in the future, as this was not the first time SpaceX has obsoleted grpc fields this project was using, nor is it likely to be the last. That can happen outside the context of this issue, though.

closed 2022-11-09 by sparky8512