sparky8512/starlink-grpc-tools #67: Error in "Running with SystemD" instructions

jbuck2005, 2022-11-10:
In the following line, I believe that there is an 's' missing from the command:
`sudo mkdir starlink-grpc-tool`

Should it not read as follows instead?
`sudo mkdir starlink-grpc-tools`

.. next

`sudo cp systemd/starlink-influx2.service /etc/systemd/starlink-influx2.service`
This did not work in Ubuntu 20.04. I had to copy the service file to /etc/systemd/**system** - perhaps the correct instruction is:

`sudo cp systemd/starlink-influx2.service /etc/systemd/system/starlink-influx2.service`

and then enable and start the scripts

sparky8512, 2022-11-10:
I didn't write those instructions, and I guess I didn't review them as closely as I should have.

I normally just drop my service conf files in the user config dir, so I'm not as familiar with where systemd keeps its system-wide conf files on various OS flavors, but it seems those instructions are not even consistent, given the following line has it in a `system` subdir, so I agree that seems to be the correct place.

jbuck2005, 2022-11-10:
resolved - thanks for the discourse :)

closed 2022-11-10 by jbuck2005