sparky8512/starlink-grpc-tools #68: Need to wrap Environment variables within /etc/systemd/system/starlink-influx2.service with " "

jbuck2005, 2022-11-10:
**Summary**:  When either the influxdb TOKEN or the ORG contain **equal (=)** signs, the script does NOT run properly

For example, my throw-away token: atIde3fpEpwT2BersHMq7iMBNOJDOfYfGXS8QDX4Lzf4c2hbTy1HwAaZSPAUWOAQH4tPT4BCIvsowoOv7DlHeA==

would break the current version of the system file due to the two equal signs at the end

**Fix**: Add **" "** to the Environment line to encapsulate the values

Running the systemd command to start logging starlink data fails with:
>Nov 10 12:30:58 hostname python3[323035]: usage: dish_grpc_influx2.py [-g TARGET] [-h] [-N] [-t LOOP_INTERVAL] [-v] [-a]
>Nov 10 12:30:58 hostname python3[323035]:                             [-o N] [-s SAMPLES] [-j] [-u URL] [-T TOKEN]
>Nov 10 12:30:58 hostname python3[323035]:                             [-B BUCKET] [-O ORG] [-k] [-C FILENAME] [-I]
>Nov 10 12:30:58 hostname python3[323035]:                             mode [mode ...]
>Nov 10 12:30:58 hostname python3[323035]: dish_grpc_influx2.py: error: SSL options only apply to HTTPS URLs
>Nov 10 12:30:58 hostname systemd[1]: starlink-influx2.service: Main process exited, code=exited, status=2/INVALIDARGUMENT
>Nov 10 12:30:58 hostname systemd[1]: starlink-influx2.service: Failed with result 'exit-code'.

My system file is:
>[Unit]
>Description=Starlink GRPC to InfluxDB 2.x exporter
>After=network.target
>[Service]
>Type=simple
>WorkingDirectory=/opt/starlink-grpc-tools/
>Environment=INFLUXDB_URL=127.0.0.1:8086 INFLUXDB_TOKEN=atIde3fpEpwT2BersHMq7iMBNOJDOfYfGXS8QDX4Lzf4c2hbTy1HwAaZSPAUWOAQH4tPT4BCIvsowoOv7DlHeA== INFLUXDB_Bucket=starlink INFLUXDB_ORG=1234b000c1234a12 INFLUXDB_SSL=false
>ExecStart=/opt/starlink-grpc-tools/venv/bin/python3 /opt/starlink-grpc-tools/dish_grpc_influx2.py -t 10 status alert_detail
>[Install]
>WantedBy=multi-user.target

When I modified the Environment statement to include " ":
>Environment=" (same as above) ''

The script runs fine.

sparky8512, 2022-11-10:
I'm no expert on systemd, but my brief testing (on Ubuntu 20.04 and 22.04) produced no problem with unquoted values with equals signs in them, and incorrect behavior when I quote the entire Environment= value.

My reading of the `systemd.exec` man page also suggests quoting the entire line is incorrect. It states the Environment= config option "Takes a space-separated list of variable assignments.", thus quoting the entire line will result in a single environment variable (the first one) being set with a value that includes the entire rest of the line after the first equals sign.

That same man page does suggest it is necessary to quote values with equals signs, though the only examples are for values that include spaces, not equals.

jbuck2005, 2022-11-11:
Ok, this definitely means some more digging. Good point about the whole string ( "  <everything here> " ) being passed as a single variable. There must be a "good" way to pass environment variables without breaking the systemd convention. I wonder if escaping the = signs (or the entire variable) in single quotes is possible ... more to investiage

I have a question about why pass variables via the environment - why not directly pass them as though you are running the script from the command prompt?

ie:

/opt/starlink-grpc-tools/dish_grpc_influx2.py -T <influx token> -B starlink -O <influx org> -u 127.0.0.1:8086 -
t 10 status alert_detail

sparky8512, 2022-11-11:
> There must be a "good" way to pass environment variables without breaking the systemd convention. I wonder if escaping the = signs (or the entire variable) in single quotes is possible ...

The (somewhat convoluted) examples in the systemd man pages have each individual environment variable assignment quoted, which I think would be:
```service
Environment="INFLUXDB_URL=http://localhost:8086" "INFLUXDB_TOKEN=<changeme>" "INFLUXDB_Bucket=<changeme>" "INFLUXDB_ORG=<changeme>" "INFLUXDB_SSL=false"
```
That syntax seems weird to me, and I would prefer to just quote the values, but since it works for me without any quotes, I don't have the ability to test whether or not that would fix whatever issue you're having.

> I have a question about why pass variables via the environment - why not directly pass them as though you are running the script from the command prompt?

I didn't write this service config file, either, but I assume it was due to a perception that environment variables would be more secure than passing on the command line.

I suppose it is marginally better, given that command lines can show up in `top` and `ps` output, while pulling environment from other processes generally requires root privilege. However, the `systemd.exec` man page specifically warns against using the Environment= option to pass secrets, given that it's exposed on D-Bus without any protection whatsoever. That would be true for the command line args, too, of course.

sparky8512, 2022-11-11:
So... I just reread your initial report, and I think I know why you were having problems. While it's true that the systemd man pages say an equals in environment variables needs to be quoted, I don't believe that's the case in modern versions of systemd.

Rather, there problem is the following variable setting (which is also a problem in the checked-in example, which I will fix):

> INFLUXDB_SSL=false

This does not do what you probably want and should just be removed. This will tell the script to use the file "false" as a CA root cert for SSL verification. You haven't specified an HTTPS URL, though, so the script is complaining about inconsistent options. The meaning of that environment variable is shared with the InfluxDB 1.x script where it made a little more sense, but the valid values are "secure", which is effectively ignored, "insecure", which is equivalent of `-I` command line option, or any other string, which is the equivalent of `-C` command line option. Whether or not to use SSL at all is determined by the URL.

I'm not entirely sure why putting quotes around the entire line worked for you. It would have stopped passing the INFLUXDB_SSL variable, thus would have avoided the specific error you were hitting, but I would have expected it to break things even worse....

Also:

> INFLUXDB_URL=127.0.0.1:8086

I would recommend using a full URL (http://127.0.0.1:8086) here rather than a bare host:port. Whether or not a bare host:port will work may be dependent on what version of the InfluxDB 2.x Python client module you have installed. (Or maybe it has always worked, and I just got confused when looking into issue #49...)

jbuck2005, 2022-11-12:
[sparky8512](https://github.com/sparky8512), I super appreciate you explaining all of your findings to me, along with the discourse on some of the "whys" like the security concerns. Thanks for taking the time to investigate this thoroughly and explain everything along the way - it helps others (such as myself) to learn, and hopefully one day contribute more effectively.

sparky8512, 2022-11-13:
No problem. There was, after all, an error in the example.

Is there anything left to resolve on this issue? If not, please close it out.

closed 2022-11-13 by jbuck2005