sparky8512/starlink-grpc-tools #72: Does not work on arm64 (aka Raspberry Pi)

DurvalMenezes, 2022-11-15:
Here's what I get when trying to run the Docker container on my RPi4b 8GB:

```
$ sudo docker run --name='starlink-grpc-tools' ghcr.io/sparky8512/starlink-grpc-tools dish_grpc_text.py -v status alert_detail
WARNING: The requested image's platform (linux/amd64) does not match the detected host platform (linux/arm64/v8) and no specific platform was requested
standard_init_linux.go:219: exec user process caused: exec format error
```

Any chance of making an arm64 version of the container available?

sparky8512, 2022-11-16:
I'll have a look. It shouldn't be a problem unless some of the required pip modules are not available for that arch.

DurvalMenezes, 2022-11-16:
> It shouldn't be a problem unless some of the required pip modules are not available for that arch.

I don't think it should be a problem, as meanwhile I installed it directly (ie outside of any container) in the same machine (following your instructions on README.md) and it's working. More specifically the `pip install [...] requirements. txt` ran with a single warning, nominally that I already had `paho-mqtt` installed.

> I'll have a look. 

Thanks! I've already started using it, but would much prefer having it run inside a container.

sponsianus, 2022-11-16:
I build ARM64 for a separate project (https://github.com/sponsianus/dishy_grafana). It's a little behind the main project but might work for your use case.

https://hub.docker.com/repository/docker/sponsianus/starlink-grpc-tools

sparky8512, 2022-11-16:
The GitHub action for publishing a multi-arch docker image looks pretty straightforward, so I will test that out in a branch.

I did test a linux/arm64 image build locally, and it built OK, but it took a couple hours because the Dockerfile has all the pip packages pinned to specific versions, some of which did not have arm64 binary wheels on pypi. It looks like later versions of those packages do, though, so I'm going to update to current versions for all of them. I probably should have been doing that periodically, anyway.

sparky8512, 2022-11-16:
OK, it has published and is tagged with the branch name, so you should be able to access it with:
```shell script
docker pull ghcr.io/sparky8512/starlink-grpc-tools:docker-arm64
```
I have only sanity tested it, and only with emulation for the ARM64 image. I will test it more thoroughly in a bit, given that I updated all the pip-installed package versions, but if someone could give the linux/arm64 image a try on an actual ARM64 device, I'd appreciate it.

acobaugh, 2022-12-01:
Note that some Raspberry Pis (mine is a 4B rev 1.1) aren't going to run an arm64 OS. Mine is `armv7l` (which looking at setup-qemu-action would be `linux/arm/v7`).

I also have to add the `pypng` package to the Dockerfile when I build it.

sparky8512, 2022-12-01:
> Note that some Raspberry Pis (mine is a 4B rev 1.1) aren't going to run an arm64 OS. Mine is `armv7l` (which looking at setup-qemu-action would be `linux/arm/v7`).

I can add a `linux/arm/v7` image easily enough, but was only planning to do so if someone requested it. The images are kinda big to be leaving around if nobody is going to use them. I take it you would use them if they were being published?

On a related topic: Is the >1GB image size going to be a problem for RPi users? Nothing in `starlink-grpc-tools` needs the full Debian base image, so the Dockerfile could be changed to use the slim version, which comes out to about 200MB. However, every time I look into doing so, I am reminded that using the full base image could actually use less storage overall if it shared with other images due to not having to add missing packages back into those other images. I don't actually use Docker much, other than testing for this project, so I don't really have an opinion on this myself.

> I also have to add the `pypng` package to the Dockerfile when I build it.

Oops, thanks for pointing that out. I guess I missed that when I added the obstruction map script. I will correct it.

acobaugh, 2022-12-02:
Yep, it's this issue: https://groups.google.com/g/grpc-io/c/vjbL3IdZ2Vk/m/EcKSeD4eAgAJ

I had to install it with the --no-binary option:
`pip3 install grpcio==1.44.0 grpcio-reflection==1.44.0 --no-binary=":all:"`

I did not try with version `1.50.0`, but it is working for me now.

sparky8512, 2022-12-21:
OK, I finally got around to actually generating the image in the main branch. I had to update my procedure for updating the "latest" tag after testing, too, but as of now, the "latest" tag points to an image that supports both linux/amd64 and linux/arm64.

If anyone wants a 32-bit ARM image to be included, please file a separate issue. Otherwise, I will look into changing the Dockerfile to add the appropriate pip flag so it works if someone makes their own image, but may or may not add it to the images posted to the repository here.

closed 2022-12-21 by sparky8512

DurvalMenezes, 2022-12-24:
> Note that some Raspberry Pis (mine is a 4B rev 1.1) aren't going to run an arm64 OS. Mine is `armv7l` (which looking at setup-qemu-action would be `linux/arm/v7`).

This, of course, depends on the specific OS. In my case, I run Devuan Chimaera 4.0 (systemd-less Debian Bullseye) and itś arm64 alright: 

    uname -a
        Linux REDACteD 5.15.74 #1 SMP PREEMPT Mon Oct 31 13:02:05 EDT 2022 aarch64 GNU/Linux

DurvalMenezes, 2022-12-24:
> OK, it has published and is tagged with the branch name, so you should be able to access it with:
> 
> ```shell
> docker pull ghcr.io/sparky8512/starlink-grpc-tools:docker-arm64
> ```

The above command is showing me the following error: 

    Error response from daemon: manifest unknown

DurvalMenezes, 2022-12-24:
> Before I do that, though, I really would like to see if anyone would actually use it.

Yep, I'm really going to use it (and BTW, sorry for the delay in coming back to this).

 I just would like to note that my architecture is arm64/aarch64 as mentioned in the issue's title.

sparky8512, 2022-12-24:
> The above command is showing me the following error:
> 
> ```
> Error response from daemon: manifest unknown
> ```

Yes, I deleted that image in order to avoid people thinking they need to use that one for ARM support. The regular one (the "latest" tag) should now be usable using the standard command from the README.

> > Before I do that, though, I really would like to see if anyone would actually use it.
> 
> Yep, I'm really going to use it (and BTW, sorry for the delay in coming back to this).
> 
> I just would like to note that my architecture is arm64/aarch64 as mentioned in the issue's title.

No problem. The 64-bit ARM image is already there now, and I was asking about the 32-bit one in that comment.

DurvalMenezes, 2022-12-24:
> The 64-bit ARM image is already there now, and I was asking about the 32-bit one in that comment.

Thanks for the clarification. 

In fact, I just `docker pull`ed it as per the README and it seems to be working perfectly.

I will keep testing it as I integrate it into my telemetry environment, and if I detect any problems I will open a more specific issue. 

Closing this one now, and thanks again,