sparky8512/starlink-grpc-tools #78: Zabbix Template for Starlink Dishy

smejdil, 2023-03-25:
Hi
I am using your project to monitor Starlink Dish using Prometeus exporter. I'm uploading templates for Zabbix 6.0

https://github.com/smejdil/zabbix-ansible/blob/main/templates/Template_Starlink_dish.xml
https://github.com/smejdil/zabbix-ansible/blob/main/templates/Template_Starlink_dish.yam
https://github.com/smejdil/zabbix-ansible/blob/main/templates/Template_Starlink_dish.jsonl

sparky8512, 2023-03-27:
Cool! Would you mind if I added links to those files on this project's Wiki (probably by expanding the scope of the [Dashboards](https://github.com/sparky8512/starlink-grpc-tools/wiki/Dashboards) page)?  ... or you could do so yourself, if you like, the page should be editable by anyone.

smejdil, 2023-03-29:
Templates can be published anywhere. It may be updated in time. So it is advisable to use my link.

sparky8512, 2023-04-08:
Thanks. I see you have updated the Wiki page so I will close this issue out.

closed 2023-04-08 by sparky8512