sparky8512/starlink-grpc-tools #81: MessageFactory class is deprecated

archerne, 2023-07-07:
when calling `starlink_grpc.status_data()` or `starlink_grpc.location_data()` a warning is produced that the MessageFactory class is deprecated.

```
.venv/lib/python3.11/site-packages/google/protobuf/message_factory.py:166: UserWarning: MessageFactory class is deprecated. Please use GetMessageClass() instead of MessageFactory.GetPrototype. MessageFactory class will be removed after 2024.
  warnings.warn('MessageFactory class is deprecated. Please use '
```

sparky8512, 2023-07-08:
I'll have a look. Probably something changed in recent version of `protobuf`. It will likely have to be addressed in `yagrc` then newer version of that pulled in.

sparky8512, 2023-07-08:
I have confirmed the behavior and opened an issue in `yagrc` to address it.

The warning appears to have been added in `protobuf` v4.23.0. It should be harmless for now, but if you really need to be rid of it before I get around to addressing this, you can downgrade `protobuf` to before that version:
``` shell script
pip install "protobuf<v4.23.0"
```

sparky8512, 2023-07-23:
This is fixed in `yagrc` v1.1.2, which has been pushed to PyPI.

I thought I would also need to update the required version of `yagrc` for this project (which would in turn require a recent version of `protobuf`), but that's not really the case: This project works perfectly fine with older `protobuf` and `yagrc` v1.1.1, it's just the later versions of `protobuf` that produce the warning when using `yagrc` v1.1.1. The pip command in this project's README will pull latest published versions of both, so will get both the `protobuf` change that added the warning and the `yagrc` change to avoid the warning.

closed 2023-07-23 by sparky8512