sparky8512/starlink-grpc-tools #82: Is it possible to poll RSSI values?

cchwala, 2023-08-05:
I am wondering if RSSI values can be polled via gRPC. I could not find it listed in the returned values that are documented in your codebase here, e.g. in [`get_status`](https://github.com/sparky8512/starlink-grpc-tools/blob/5cbc5b73dda7c32531620dbc4e79376ffcc85a87/starlink_grpc.py#L668). But I got info from somebody that RSSI is shown in the "command debugging":

[image]

Hence, I assume it can be polled somehow. Could RSSI be polled via the starlink gRPC tools?

sparky8512, 2023-08-06:
There's nothing that looks like RSSI in the dish status response gRPC message.

There are a couple fields that look like RSSI values of some kind in the TransceiverGetTelemetryResponse message, which is the response type of the `transceiver_get_telemetry` request. However, when I try to issue that request to my dish, I get an RPC error with status code UNIMPLEMENTED. That's been my experience for most of the requests that have the more technical detail: they return an error claiming either unimplemented or unauthorized status.

There are a number of messages with WiFi related RSSI values, but those are only relevant to the router gRPC, not the dish gRPC. The values in the screen cap you posted look like they are either from one of those or from the local device's WiFi status (meaning it would be about the phone's view of the WiFi connection).

cchwala, 2023-08-06:
@sparky8512 Thanks for the swift reply!

It seems like it is currently not possible to get the RSSI values from the dish. I also suspected that the RSSI value shown in the screenshot could be from the Wifi, not the dish.

Given that it was disabled to get SNR values, it does look like there is currently no way to get information on the received signal level of the dish, which is unfortunate, because it could provide important information about the state of the atmosphere, e.g. about rainfall rate along the path between dish and satellites.

If there is no further info on that, feel free to close this issue.

closed 2023-08-06 by sparky8512