sparky8512/starlink-grpc-tools #85: Obstruction map - which way is "up" ?

iam-TJ, 2023-10-15:
Late to the party but wondering how to map compass directions to the PNG ?

I'm tending to think the lower edge of the PNG is South?
[image]

My dish has unobstructed 360 view but there is a tall structure to the north ("behind" the dish). Latitude is 54 degrees.

sparky8512, 2023-10-15:
I guess I never did document that, probably because the protocol doesn't indicate what the individual map points mean beyond "rows" and "columns". The way the mobile app display that map data makes it pretty apparent, though.

The exact technical description is a little involved due to coordinate system being converted into Cartesian coordinates, but if you only care about azimuth (compass direction), that's simple enough: The center top of the image is North, top right corner is Northeast, etc.

Altitude (elevation) angle is a little more complicated. The center of the image is "up" (zenith), and it gets closer to the horizon the further away from the center point it is. Note that this is center of the image, not center of the blob of white pixels, and "up" is from the perspective of an observer located at the dish, not the direction the dish is pointing.

sparky8512, 2023-10-29:
There is now a more technical description of direction to pixel coordinate mapping documented at the top of the `dish_obstruction_map.py` script. 

If there are no further questions, let's close out this issue.

closed 2023-10-29 by sparky8512