sparky8512/starlink-grpc-tools #86: Get sleep mode status?

boswelja, 2023-10-25:
I see we can set the sleep state now, awesome! Do we have a means of retrieving the current sleep state?

For context, I'm looking at adding these controls to HomeAssistant, it'd be cool if dishy was the source of truth for the data (otherwise changing settings in the Starlink app wouldn't reflect in HASS)

boswelja, 2023-10-26:
> If you're looking for an indication of whether or not the dish is currently in sleep state, I think that may be reflected in the alerts (as `alert_is_power_save_idle`, probably), but I don't have access to my dish at the moment to verify.
> 
> If you're looking for the current settings, you can get them via:
> 
> ```
> python dish_control.py set_sleep
> ```
> 
> with no additional arguments.

I'm after the latter, getting the current sleep settings.

Does the same apply to the `starlink_grpc_core` library? I see a `set_sleep_config` that returns `None` https://github.com/sparky8512/starlink-grpc-tools/blob/5cbc5b73dda7c32531620dbc4e79376ffcc85a87/starlink_grpc.py#L1538

sparky8512, 2023-10-26:
> Does the same apply to the `starlink_grpc_core` library?

Hmm... it appears not. `dish_control` does not use the core library, but I try to keep the functionality in parity between them. It looks like I missed adding get configuration to the library.

I will add that, as a new function, probably some time next week.

sparky8512, 2023-10-29:
That change adds the new function `get_sleep_config` to the core module.

I still need to push a new version tag to get that reflected in the pip-installable core package.

boswelja, 2023-10-29:
Nice, thanks!

sparky8512, 2023-10-30:
This has now been published to PyPI as v1.1.3. I believe that should close out this issue.

closed 2023-10-30 by sparky8512