sparky8512/starlink-grpc-tools #87: Docker image: Failed writing to InfluxDB

nkinnan, 2023-12-24:
I'm attempting to debug an issue with a HomeAssistant plugin based on starlink-grpc-tools.  As a first step I downloaded the Docker image for starlink-grpc-tools:v1.1.2 to see what it was reporting.  Unfortunately I can't make any progress due to this error:

> ERROR: Failed writing to InfluxDB database: HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /write?db=starlinkstats&precision=s (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fe03c6bbcd0>: Failed to establish a new connection: [Errno 111] Connection refused'))

This is with Docker on Windows 10, under both WSL and Hyper-V back-ends.  

Strangely when running under WSL I got this error immediately on first container start.  When running under Hyper-V it seemed to work the first time except that my starlink was turned off so the image exited with this message:

> ERROR: Dish unreachable and ID unknown, so not recording state

Turning the starlink back on and trying to run the image again, I get the same error about InfluxDB.  

> ERROR: Failed writing to InfluxDB database: HTTPConnectionPool(host='localhost', port=8086): Max retries exceeded with url: /write?db=starlinkstats&precision=s (Caused by NewConnectionError('<urllib3.connection.HTTPConnection object at 0x7fe03c6bbcd0>: Failed to establish a new connection: [Errno 111] Connection refused'))

The Docker back-end being WSL or Hyper-V may or may not be related to the issue but I try to be thorough.  It stuck out at me that the Hyper-V version did seem to run successfully once before reverting to the same behavior as when using the WSL back-end.

The installation command line I used per the documentation (I ran this twice, since apparently Docker on Windows either deletes images when switching back-ends, or doesn't share them):

docker pull ghcr.io/sparky8512/starlink-grpc-tools:v1.1.2

Please let me know anything I can do to assist such as pulling logs, etc.

sparky8512, 2023-12-24:
It sounds like you have this integrated with an InfluxDB server running somewhere, and the problem is that the scripts cannot connect to it.

> When running under Hyper-V it seemed to work the first time except that my starlink was turned off so the image exited with this message:
> 
> > ERROR: Dish unreachable and ID unknown, so not recording state
> 
> ...
> 
> It stuck out at me that the Hyper-V version did seem to run successfully once before reverting to the same behavior as when using the WSL back-end.

If the error about dish unreachable is what you mean when it ran successfully once, that would not have indicated a success of the InfluxDB part. If the dish is unreachable from the very first sample, the scripts won't know the dish ID, so it won't even try to record anything to the InfluxDB database, not even the fact that the dish was unreachable. This is because it uses the dish ID as a key in the database, so if it doesn't know that, it cannot write anything to the database.

The fact that you get an InfluxDB error when your dish is powered on indicates that the scripts are able to get data from the dish OK, and the problem is with the InfluxDB server. The "Connection refused" part of the error message means that the port the script is trying to talk to is not open. Either the InfluxDB server is not listening on that port or it's being blocked by some firewall. If your InfluxDB server is being run in a container, it may be that the port is not being forwarded into the container.

nkinnan, 2023-12-24:
> It sounds like you have this integrated with an InfluxDB server running somewhere

I'm new to docker and running containers, the only thing I've done so far is install docker, pull starlink-grpc-tools:v1.1.2 via the command line per the instructions, and try to start the container in docker desktop.  I've never used an InfluxDB so I assumed this was something the container was pre-configured to do.  Are you saying it's not standard for the container to attempt to connect to an InfluxDB instance?  If so, I'm confused because I made no changes at all to the container, I just hit run.

> If the error about dish unreachable is what you mean when it ran successfully once

Yes, that was poor wording on my part.  I should have said it was the only time I didn't see the error about InfluxDB, and the image continued to run for a minute or so before exiting.  When I got the InfluxDB error the image exited immediately.

Really, all I want to do is get to a console interface and examine the output of the scripts, but I'm unsure how to proceed since the container immediately exits with the InfluxDB error.

nkinnan, 2023-12-24:
I think I just realized what happened.  Going back to the packages page, there are instructions I completely missed.  Reading through those:

> The default command is dish_grpc_influx.py status alert_detail, which is really only useful if you pass in environment variables with user and database info

So that explains it, and I should be using the command line to start the container and specify which script to run.  My mistake! 
 Apologies for completely missing those instructions before now.

I did hit another small hiccup, this command line as suggested did not work:

> docker run --name='starlink-grpc-tools' ghcr.io/sparky8512/starlink-grpc-tools dish_grpc_text.py -v status alert_detail
> docker: Error response from daemon: Invalid container name ('starlink-grpc-tools'), only [a-zA-Z0-9][a-zA-Z0-9_.-] are allowed.

It seems that on windows, the single quotes around the name argument should be omitted.  This did work:

> docker run --name=starlink-grpc-tools ghcr.io/sparky8512/starlink-grpc-tools dish_grpc_text.py -v status alert_detail

So it might be good to note this in the instructions for anyone else running on windows.  

In any case, this can be closed, thanks!

nkinnan, 2023-12-24:
I think I just realized what happened.  Going back to the packages page, there are instructions I completely missed.  Reading through those:

> The default command is dish_grpc_influx.py status alert_detail, which is really only useful if you pass in environment variables with user and database info

So that explains it, and I should be using the command line to start the container and specify which script to run.  My mistake!  Apologies for completely missing those instructions before now.

closed 2023-12-24 by nkinnan