sparky8512/starlink-grpc-tools #94: Power consumption?

lazmo88, 2024-09-24:
Starlink has added power consumption to their app, assuming this is also available through grpc. Do we need to update the code to get access to new data?

https://github.com/home-assistant/core/issues/125526

sparky8512, 2024-09-24:
I guess they added that to the history data about a month ago. I'll have a look at adding it. Will have to think about how this affects backwards compatibility for the bulk history data, though.

sparky8512, 2024-09-24:
Hmm.... On my dish, the new field just reports an array of 0.0 values, and it does not show up in the app that I could find. I'm guessing this functionality is not supported by the original (round) dish hardware. I should still be able to add support for this, but I'll need some help testing.

Also, if someone with a dish that supports this could post the results of the following python commands, that would help me understand the units used so that I can properly document it:
```python
import starlink_grpc
print(starlink_grpc.get_history().power_in)

sparky8512, 2024-09-25:
I have checked in support for this into a new power-stats branch. As I mentioned above, this does not work on my dish, so I have done only light testing on it. In particular, I am making an assumption that the data is reporting in watts, which is just an educated guess.

If someone with a dish that supports this functionality can test that what it spits out looks correct, and that the values look correct for being watts, then I will merge it into the main branch and spin a version number for it, so I can update the core module package on PyPI. General details are in the change description posted above and specific details are in the docstring at the top of `starlink_grpc.py`.

trinler007, 2024-09-25:
It works on my starlink mini. If've attaced the output of

```
import starlink_grpc
print(starlink_grpc.get_history().power_in)
```
Hope that helps...
[power_in.txt](https://github.com/user-attachments/files/17136677/power_in.txt)

sparky8512, 2024-09-25:
Thanks. That should do.

sparky8512, 2024-09-27:
OK, change has been merged to main branch and is tagged as release v1.2.0.  See release notes for detail on other changes since prior release.

The v1.2.0 core module has also been posted to PyPI and the docker image has been updated.