sparky8512/starlink-grpc-tools #96: Full tools release on Pypi?

Bluemethyst, 2024-09-26:
Not an actual issue, I'm using the [core release from pypi](https://pypi.org/project/starlink-grpc-core/) for some projects and I'm wondering if you can update the pypi version to include all the functions and data capturing methods? i would like to be able to see devices on the network and power consumption data from that module for example. No problem if you're busy or don't see value in it :)

sparky8512, 2024-09-26:
I'm still working through full testing on the power consumption data, I should get that finished up and published within a few days. I think that's the only think that has not yet been posted to the PyPI package.

I'm unclear on what else you're asking for. Are you looking to have the scripts that push the data into the various data backends published? I haven't done that to date because the script naming is a little awkward to be at top-level and nobody has requested it so far, so I decided not to worry about trying to move them into a package namespace.

If you're looking for some other functionality to be supported by the core module, you'll have to be more specific. The devices on network thing you mentioned sounds like a function of the router, which these scripts are not currently targeting.

Bluemethyst, 2024-09-26:
hmm okay maybe I am thinking more router controls then, I saw there was one script that allowed us to reboot the starlink, stow it, etc. basically the dish_control.py and dish_check_update.py from a quick look. and is it possible to implement the router controls/communication? or is that a whole different thing?

sparky8512, 2024-09-27:
The Starlink router does also use grpc for most, if not all, of whatever you can configure for it in the Starlink mobile app, so if you can do it in the app, it should be possible to do it with a script.

The grpc protocols the router uses are even shared with the dish. However, most of the requests are either dish-specific or router-specific, so they might just as well be separate. The router-specific requests tend to change all the time, though, whereas the dish-specific requests change very slowly. In practice, this means it's not practical to try to keep up with all the latest changes on the router side. I would expect that basic configuration and control requests, like the ones you mentioned, are probably stable enough, though.

There are 2 main reasons I have not messed with the router grpc to date:
1) The dish scripts were originally about polling and recording information, not about configuration or control. It didn't seem to me that there was much point recording the information that could be polled from the router.
2) I don't use the Starlink router myself and it's been well over 3 years since I even had it plugged in.

I can see the usefulness of configuration and control scripts for the router, so I think that first reason is no longer valid, but the second is still very much in play.

Bluemethyst, 2024-09-27:
okay no problem, thanks so much for your time. Could you maybe point me towards some resources for finding out how i could interface with the router myself as i am still using my Gen2 router with my dishy?

sparky8512, 2024-09-27:
As far as I know, SpaceX has not published the protocol (`.proto`) files publicly. Which means that posting the generated protocol modules, such as the one you pointed to, is probably trampling on SpaceX copyright (at least in the US), as they are clearly derived from the reflection data. SpaceX may or may not actually care about that, but this is the reason why I don't ever post them, and primarily use reflection at run time to get the Python grpc data structures.

The dish has the grpc reflection service enabled, though, and I expect the router does, too. As such, it is easy enough to browse the protocol using grpcurl (which can be found [here](https://github.com/fullstorydev/grpcurl)). I wrote up instructions for doing that on [a Wiki page](https://github.com/sparky8512/starlink-grpc-tools/wiki/Useful-grpcurl-commands#inspecting-the-protocol). Those instructions were written for the dish, but should work on the router IP, too. According to [my notes](https://github.com/sparky8512/starlink-grpc-tools/wiki/TCP-Ports-Used-by-the-gRPC-Service), the router uses port 9000 for this instead of 9200 that the dish uses.

sparky8512, 2024-11-20:
I'm going to close this issue. Feel free to re-open, or to open a new one, if you have further questions.

closed 2024-11-20 by sparky8512