sparky8512/starlink-grpc-tools #101: Is it possible to to modify the setting "use starlink positioning exclusively"?

prtngn, 2024-12-01:
Hello. Thank you for this utility. I have a question: is it possible to manage the “use starlink positioning exclusively” setting, which is located in the Debug Data section of the application, through the gRPC protocol?

I live in a region where Starlink has not been officially launched yet, but it’s available in a neighboring country. As a result, I have to manually activate this setting after each restart. I’d like to automate this process.

sparky8512, 2024-12-01:
Yes, this should be possible. As far as I can tell, this setting flips the "inhibit GPS" flag on the dish, and the request for that does not appear to require authentication. This can be added to `dish_control.py` easily enough.

sparky8512, 2024-12-03:
That change should add the functionality you want. The following should be the equivalent of turning the "Use Starlink positioning exclusively" toggle switch on:
```shell script
python dish_control.py set_gps --no-enable
```
and toggling it off would be:
```shell script
python dish_control.py set_gps --enable
```

Still to be done:
- add loop control to that script in order to facilitate automation without having to go through `cron` or similar
- add the GPS enablement state to the status polling data, either in `status` or a new data group
- update README to document usage

sparky8512, 2024-12-04:
All changes are now in.

closed 2024-12-04 by sparky8512

prtngn, 2024-12-05:
Thank you very much for the edits. Everything works.