sparky8512/starlink-grpc-tools #102: Is it possible to track the current connected satellite id?

ItakEjgo, 2024-12-03:
Hi,
I am wondering whether it is possible to identify the satellite which is currently connected.
It seems that there was a call _dish_get_context_ which can provide the satellite id, but this call is unavailable now.

sparky8512, 2024-12-03:
I'm pretty sure the short answer is no.

The `dish_get_context` request is still there, but it returns an authorization error when run locally. I'm not even sure it has the actual current satellite ID, anyway. There is also a `transceiver_get_telemetry` request that does have what might be current satellite ID, but that also returns an authorization error when run locally. It could be that these can only be run remotely, or they might be locked down in some other way.

I suppose it would be technically possible to track the satellites by resetting the obstruction map, examining where new map data fills in, and correlating that with satellite position data, but doing so would require periodically resetting the map and that would probably impact the dish's ability to avoid obstructions.

ItakEjgo, 2024-12-03:
> I'm pretty sure the short answer is no.
>
> The `dish_get_context` request is still there, but it returns an authorization error when run locally. I'm not even sure it has the actual current satellite [...]

Thanks for your reply!

closed 2024-12-03 by ItakEjgo