sparky8512/starlink-grpc-tools #103: starlink_grpc.py crashes - latency update

cluz, 2024-12-03:
Hello,

First thanks for your work, I've been using your scripts for more than one year to monitor several starlinks and it works super fine.

I've recently done the update to v1.21 to solve the issue with latency that was no more retrieved but since the program crashes after several hours with the following error:
starlink_grpc.py", line 785, in status_data
latest_index = latest_range[0]
TypeError: 'itertools.chain' object is not subscriptable

Note that I'm using python 3.8 with a custom version derived from dish_grpc_text.py with arguments status alert_detail usage power -s 60 -v
The only modification that I've done was the update of starlink_grpc.py

Thanks for your help.

sparky8512, 2024-12-03:
This looks like it's due to a corner case where the history buffer wraps, which means it could happen any time one of the status-derived data groups is pulled exactly at that wrap point (which happens every 15 minutes). This is bad, as that can affect anyone who uses that version. I will try to get a fix out today.

sparky8512, 2024-12-03:
Fix has been released as v1.2.2. Updated core module has been published to PyPI and docker image has been updated.

Thanks for filing this, as I would not have noticed it otherwise, since I only poll history stats, not status.

closed 2024-12-03 by sparky8512

cluz, 2024-12-04:
Thanks for your reactivity, I have applied the update and so far it looks ok