sparky8512/starlink-grpc-tools #104: Wifi bypass

sudanking, 2024-12-10:
Can I enable wifi bypass mode through grpc?

sparky8512, 2024-12-10:
I assume so. There's a `wifi_setup` request, which has a `bypass` boolean in it. As far as I know, this setting persists across reboots, so I'm guessing most people just use the Starlink app to set it and then don't have to mess with the Starlink router after that.

There is also a `wifi_set_config` request which looks like it has many more setting in it, some of which look like they probably need authorization, so may not work via local grpc (or at all).

I can't really comment much more than that, as I don't use the Starlink router myself. Even if I did, I have the original round dish, which can be used without the router entirely, so no need to bypass. If you want to poke around this yourself, you can do so most easily using [gprcurl](https://github.com/fullstorydev/grpcurl). I posted some notes about how to do that [here](https://github.com/sparky8512/starlink-grpc-tools/wiki/Useful-grpcurl-commands#inspecting-the-protocol) but see also [here](https://github.com/sparky8512/starlink-grpc-tools/wiki/TCP-Ports-Used-by-the-gRPC-Service) for the different port number needed to examine the router.

sparky8512, 2025-01-22:
I'm going to assume there are no further questions here and close this issue out.

closed 2025-01-22 by sparky8512