sparky8512/starlink-grpc-tools #106: Can i controle connected users using grpc-tools

Ayman-Nageeb, 2024-12-21:


sparky8512, 2024-12-22:
If you mean WiFi connected users, then no, not with these tools. The scripts in this project do not operate on the router, only on the dish.

Ayman-Nageeb, 2024-12-22:
Is

> If you mean WiFi connected users, then no, not with these tools. The scripts in this project do not operate on the router, only on the dish.

Is there any tools to control the router ?  (install OpenWRT) or any other way ?

sparky8512, 2024-12-22:
I'm not aware of any such tools, but there may be. I haven't been keeping up to date on what people are doing with the Starlink router, because I don't use it myself.

If you want full router control, I recommend putting the Starlink router in bypass mode (or just remove it if you have a dish version where that is an option) and use your own router.

sparky8512, 2025-01-22:
I'm going to assume there are no further questions here and close this issue out.

closed 2025-01-22 by sparky8512