sparky8512/starlink-grpc-tools #113: FEATURE: Set "Flat Mode" and "Atuomatic" in Tilt settings

askpatrickw, 2025-02-23:
I'm not sure yet how to fix this on my own, but I'm working it out...  PR to come, advice welcome

The mobile app supports setting the Starlink to flat mode.

[image]

I can see this in the dish_get_config.
 
```cli
grpcurl -plaintext -d '{ "id": 1, "dish_get_config": {} }' 192.168.100.1:9200 SpaceX.API.Device.Device.Handle

{
  "id": "1",
  "apiVersion": "30",
  "dishGetConfig": {
    "dishConfig": {
      "snowMeltMode": "ALWAYS_OFF",
      "levelDishMode": "FORCE_LEVEL",
      "swupdateRebootHour": 3,
      "applySnowMeltMode": true,
      "applyLocationRequestMode": true,
      "applyLevelDishMode": true,
      "applyPowerSaveStartMinutes": true,
      "applyPowerSaveDurationMinutes": true,
      "applyPowerSaveMode": true,
      "applySwupdateThreeDayDeferralEnabled": true,
      "applyAssetClass": true,
      "applySwupdateRebootHour": true
    }
  }
}
```

askpatrickw, 2025-02-23:
Seems like all you need to is include (or not) `"levelDishMode": "FORCE_LEVEL"`

sparky8512, 2025-02-23:
Unfortunately, I don't think the ability to set this will be doable with the scripts in this project as they are now. The `dish_get_config` request will work fine to query the current setting, but changing it requires the `dish_set_config` request.

When I try `dish_set_config` on my dish, I get a `PERMISSION_DENIED` RPC error, even when I don't set any of the "apply" members to true. I'm pretty sure this means this request cannot be performed via local grpc, and instead must be done remotely. If the app requires you to sign in to your account in order to change this setting, that would be further evidence that this is the case.