sparky8512/starlink-grpc-tools #115: No module named 'spacex'

parsley72, 2025-05-15:
I tried running but get:
```
$ python3 dish_grpc_text.py
Traceback (most recent call last):
  File "dish_grpc_text.py", line 21, in <module>
    import dish_common
  File "/home/l6a/dish_common.py", line 23, in <module>
    import starlink_grpc
  File "/home/l6a/starlink_grpc.py", line 414, in <module>
    from spacex.api.device import device_pb2
ModuleNotFoundError: No module named 'spacex'
```
This isn't in requirements.txt, where is it supposed to come from?

sparky8512, 2025-05-15:
If you have `yagrc` installed (which is in requirements.txt), it should pull it from the dish using reflection. At that point in the file, the actual reflection part is still pending, and `spacex.api.device` is just on a list of modules that should allow import for later resolution.

If you don't want to use reflection at script run time, you can generate the `spacex` modules ahead of time using [these instructions](https://github.com/sparky8512/starlink-grpc-tools/wiki/gRPC-Protocol-Modules), but I don't recommend that unless you have some specific reason for it.

If you do have `yagrc` installed, but you're still getting that error, then something is going wrong with `yagrc`, possibly in conjunction with a new Python version, as I haven't been keeping up with testing latest versions of that.

parsley72, 2025-05-15:
My bad, I'd missed `yagrc`. All working now.